pcbway/PCBWay-Plug-in-for-Kicad #2: File save loaction

jonaslee228, 2022-05-16:
Could you have a file explorer come up and ask where you want to save the .zip file? Instead of a temp file.
We keep the .zip file in each iteration for our records. Also the ability to name the .zip file to the project name and version of the board. Maybe use the variables supplied by Kicad?

PHIL-IP, 2022-11-22:
Hi,
naming the file with the project name instead of a temporary file name would indeed be much better and would allow us to better track our orders

pcbway, 2022-11-22:
Thanks for the feedback,I have told to our engineer team and will update it

jonaslee228, 2023-01-25:
Did this ever get done? It would save us a lot of time.

KanyonKris, 2023-02-15:
Another vote for this feature.

KanyonKris, 2023-02-15:
Version 0.1.3 uses the PCB filename instead of a random generated name. This is good enough for me.