pcbway/PCBWay-Plug-in-for-Kicad #6: Not working with recent nightly 6.99

Robyrob, 2022-08-16:
Using it with recent nightly release, it stop working.
I think because kicad programmer changed the option "SetExcludeEdgeLayer()

pcbway, 2022-08-16:
Hi,I didn't get the meaning.
We are in China time and just tried,it is workig well.
Could you tell me the details?

Robyrob, 2022-08-16:
I am using the nightly version of Kicad:
Version: (6.99.0-2815-gbbc0c61ccb), release build in Windows 64
In stable version of kicad 6.0 Your plugin is working; in nightly version it doesn't work. The procession bar stop and the files are not generated.
In the python console the message is:'PCB_PLOT_PARAMS' object has no attribute 'SetExcludeEdgeLayer'

pcbway, 2022-08-16:
Thanks,I will contact with KiCAD team about this.

sethhillbrand, 2022-11-02:
The line 

```python
        popt.SetExcludeEdgeLayer(True)
```

Should be

```python
        if hasattr(popt, "SetExcludeEdgeLayer"):
            popt.SetExcludeEdgeLayer(True)
```

pcbway, 2022-11-03:
Thansk for the help,last time we didn't find the way out

greggjaskiewicz, 2023-01-25:
it doesn't work wth the 7.0rc1 still. Please make sure it does, because many of us are using it and trying to push designs using this plugin.

jonaslee228, 2023-01-25:
Yes I agree. It is very unhandy to have this not working. Please fix it soon.

subst4nc3, 2023-01-26:
KEEP IN MIND THAT I'M NOT A PROGRAMMER....
so the modification are proposed without any warranty and i'm not responsible if you receive an incorrect board made with these modifications.

Hello, i have managed to make it working again by changing and adding some lines in the thread.py file, according to the changes made to another similiar plugin (AISLER):

modify from:
         boardWidth = pcbnew.Iu2Millimeter(board.GetBoardEdgesBoundingBox().GetWidth())
         boardHeight = pcbnew.Iu2Millimeter(board.GetBoardEdgesBoundingBox().GetHeight())
to:
         boardWidth = pcbnew.ToMM(board.GetBoardEdgesBoundingBox().GetWidth())
         boardHeight = pcbnew.ToMM(board.GetBoardEdgesBoundingBox().GetHeight())

----add:
            angle = f.GetOrientation()
            try: # kicad >= 6.99
                angle = angle.AsDegrees()
            except AttributeError: # kicad <7
                angle /= 10.0
---before components.append({ statement.

modify from:
                'rotation': f.GetOrientation() / 10.0,
to:
                'rotation': angle,

pcbway, 2023-02-15:
Now we have solved the problem and updated the plugin that suitable for version7

greggjaskiewicz, 2023-02-15:
still not available in Kicad itself tho.

KanyonKris, 2023-02-15:
Version 0.1.3 works for me in KiCad 7. Downloaded the plugin and installed manually. I assume version 0.1.3 will be added to the official KiCad plugin repository eventually.

pcbway, 2023-02-16:
We will add to the official KiCad plugin repository in gitlab soon