pcbway/PCBWay-Plug-in-for-Kicad #8: BOM and centroid file generation

rhard, 2022-10-18:
It looks like the `PCBWay_components.json` is not accepted for assembly projects. The review team asked me to provide a standard BOM and centroid files as usual. At least, this is the experience I have.

Is it possible to generate those files by the plugin?

pcbway, 2022-10-19:
Thanks for the reply. We are now updating the order system,so you could place assembly order directly in the future.
Sorry for the inconvenience at the moment.

rhard, 2022-10-19:
Meanwhile, if the order system update will take a long time, you can check the JLC plugin. It can export the BOM and POS files with slightly modified code: https://github.com/bennymeg/JLC-Plugin-for-KiCad/blob/master/plugins/process.py

pcbway, 2022-10-19:
We could get all files,just in the json format.So the sales not familiat with it. The order system will update soon.

brodiefairhall, 2023-02-07:
Is there any update as to when the sales team will be able to use the JSON file for assembly?
I had an order sitting for weeks that I didn't realise was because the sales team could not use the JSON file in the export.

pcbway, 2023-02-08:
Now the sale could use the JSON file to place assembly order. And we will update plugin soon