pcbway/PCBWay-Plug-in-for-Kicad #9: Export doesn't support the “Exclude from BOM” and “Exclude from position files” options

rhard, 2022-10-19:
Please support the above-mentioned options so the BOM will not be populated with unneeded components. You can check the JLC repository for a solution: https://github.com/bennymeg/JLC-Plugin-for-KiCad/blob/master/plugins/process.py

```python
 if not footprint.GetAttributes() & pcbnew.FP_EXCLUDE_FROM_POS_FILES:
```

```python
 if not footprint.GetAttributes() & pcbnew.FP_EXCLUDE_FROM_BOM:
```

[image]

accsysorg, 2023-11-07:
I had the same issue, and have fixed it. I can't push the code change, but you just need to add the following at line 111 in process.py:

`if placed == True:`

Then include the for and if statement underneath in the conditional block.

chibiconsulting, 2024-09-02:
Still an issue in KiCad v8.0.4 with plugin v1.0.0.

pcbway, 2024-09-03:
Could you tell me the detail issue?

chibiconsulting, 2024-09-03:
Sure, see T-1P17W321128A. Simple circuit, LED, Resistor & Pin header.
Pin header has "Exclude from bill of materials" checked.
[image]
Pin header is included in BOM:

Designator | Quantity | Value | Footprint | Package
-- | -- | -- | -- | --
D1 | 1 | LED | LED_0603_1608Metric | smt
**J1** | **1** | **Conn_01x02** | **PinHeader_1x02_P2.54mm_Vertical** | **tht**
R1 | 1 | 100 | R_0603_1608Metric_Pad0.98x0.95mm_HandSolder | smt

pcbway, 2024-09-03:
Hi
I get it,thanks very  much. I have asked the engineer evalute it.
Now if you choose DNP,it will show in the bom.
But if choose exclude from bill of materials,it will not show it.Thanks for the feedback.

pcbway, 2024-09-05:
Before we fix the problem,you could choose DNP,so our sale will not make quotation for the component