pcbway/PCBWay-Plug-in-for-Kicad #10: Does the plugin support boards with more than 4 copper layers?

evantandersen, 2022-12-27:
I have a 6 layer design, I tried the pcbway plugin, it only uploaded 4 of the copper layers to PCBway. Luckily the staff caught the mistake on design review. I exported the gerbers/drill files manually and got all 6 layers.

pcbway, 2022-12-27:
Hi,Thanks for feedback. It is strange as the layer parameter was passed by Kicad.
Could you tell me your order number? I will ask the engineer check it.

evantandersen, 2022-12-27:
W549251AS2P3. It 'detects' 6 layer PCB. BUT, the uploaded zip only contains the first two inner copper layers, In1.Cu and In2.Cu. In3.Cu and In4.Cu are missing.

evantandersen, 2022-12-27:
For reference, when i manually export gerbers/drill from KiCAD, I got the files on order W549251AS2P2

pcbway, 2023-02-15:
Now we updated the plugin for version7,please double check it

EverythingSmartHome, 2024-03-20:
I am having this issue with an 8 layer design, the plugin generates 6 layers instead of the correct 8 layers. Generating the gerbers manually correctly generates all 8 layers.

pcbway, 2024-03-25:
Hi,Could you send the design file to anson@pcbway.com? Our engineer team will test it

wlangeveld, 2024-09-13:
Just used it with 10 layer board. It only submits the first 6 layers. This is with the latest version of KiCAD 8.

pcbway, 2024-09-17:
Could you send the original kicad file to anson@pcbway.com.we will check it.
Regards