pcbway/PCBWay-Plug-in-for-Kicad #11: PCB Assembly file is not correct:

AlaaSaadAbdo, 2023-02-20:
Hi,

I am using Kicad version 7.
I am using the plugin with the latest version 1.3.

When I try to export the PCBA JSON file there are some mistakes according to my understanding.

Screenshot of schematic bom table:
[image]

Screenshot of the json generated without `mpn` and the `pack` is wrong.
[image]

Cheers and thank you for the plugin

khill25, 2023-04-24:
+1 to the issue, I am seeing the same thing using Kicad 6.

pcbway, 2023-04-25:
Hi，I will ask the engineer check it soon