pcbway/PCBWay-Plug-in-for-Kicad #13: Gauge.SetValue(): argument 1 has unexpected type 'float'

CarlFK, 2023-04-18:
https://github.com/pcbway/PCBWay-Plug-in-for-Kicad/blob/daf97d0544836c2f16882cf09d18ac330ac167f7/plugins/plugin.py#L43
```
Traceback (most recent call last):
  File "/home/carl/.local/share/kicad/7.0/3rdparty/plugins/com_github_pcbway_PCBWay-Plug-in-for-Kicad/plugin.py", line 43, in updateDisplay
    self.m_gaugeStatus.SetValue(status.data)
TypeError: Gauge.SetValue(): argument 1 has unexpected type 'float'
(repeat 100's of times)
```
KiCad help, about, copy version:
```
Application: KiCad PCB Editor x86_64 on x86_64
Version: 7.0.2-6a45011f42~172~ubuntu22.04.1, release build
Libraries:
    wxWidgets 3.2.1
    FreeType 2.11.1
    HarfBuzz 6.0.0
    FontConfig 2.13.1
    libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.14
Platform: Ubuntu 22.04.2 LTS, 64 bit, Little endian, wxGTK, ubuntu, wayland
Build Info:
    Date: Apr 17 2023 07:58:00
    wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
    Boost: 1.74.0
    OCC: 7.5.2
    Curl: 7.88.1
    ngspice: 38
    Compiler: GCC 11.3.0 with C++ ABI 1016
Build settings:
    KICAD_SPICE=ON
```

honeybadger2023, 2023-05-17:
I had the same issue and the files uploaded to PCBway somehow missed the stitched vias. The drillpattern was there but the small vias where missing. Creating the files manually and uploading them worked and showed the stitching.

pcbway, 2023-05-17:
> I had the same issue and the files uploaded to PCBway somehow missed the stitched vias. The drillpattern was there but the small vias where missing. Creating the files manually and uploading them worked and showed the stitching.

Hi,could you send the file to me? I will ask the engineer test it and try fo find the code problem.

honeybadger2023, 2023-05-17:
Hi, sure. Can you provide an Email?

pcbway, 2023-05-17:
> Hi, sure. Can you provide an Email?
Please send to anson@pcbway.com

honeybadger2023, 2023-05-17:
Files are on their way.

pcbway, 2023-05-24:
We update the plugin and solve the problem,also we will allow place assembly order  directly soon.