pcbway/PCBWay-Plug-in-for-Kicad #14: Kicad Nightly build plugin not working

robouden, 2023-08-08:
Trying to export of the Gerber files for the latest KiCad build and the PCB way plugin keep crashing. System is Linux Mint 21.1
I tried to modify the code as suggested by others in older issues, but no luck.

Regards
Rob Oudendijk

a-allard, 2024-01-14:
Got this working.....

To modify this to work with Kicad 8.0 open process.py and find the function `get_mpn_from_footprint`

replace with this modified version:

```python
    def get_mpn_from_footprint(self, f):
        keys = ['mpn', 'MPN', 'Mpn', 'PCBWay_MPN' ,'part number', 'Part Number', 'Part No.', 'Mfr. Part No.', 'Mfg Part']
        for key in keys:
            if f.HasField(key):
                return f.GetFieldByName(key).GetText()
````

pcbway, 2024-01-15:
Thanks for the feedback,I will ask our engineer double check it

a-allard, 2024-02-09:
@pcbway Any update on when this might get fixed?

a-allard, 2024-02-09:
If it helps I can do a fork and pull request....

pcbway, 2024-02-23:
Hi
There are some test issues in kicad8 night version that have not been solved yet. You can use kicad7 version first.
Sorry about that

aldwinhermanudin, 2024-02-28:
> Got this working.....
>
> To modify this to work with Kicad 8.0 open process.py and find the function `get_mpn_from_footprint`
>
> replace with this modified version:
>
> ```python
>     def [...]

I'm using the latest KiCAD version 8.0.0. This changes work for me and the output files looks good too.

pcbway, 2024-02-28:
If by this method, the kicad7 version will not be able to use the plug-in. We are communicating with the kicad manager about this.

a-allard, 2024-02-28:
@pcbway That is a fair point....

Here is a fix that should work for either 7 or 8.  (I have not tested yet)

```python
def get_mpn_from_footprint(self, f):
        keys = ['mpn', 'MPN', 'Mpn', 'PCBWay_MPN' ,'part number', 'Part Number', 'Part No.', 'Mfr. Part No.', 'Mfg Part']
        v = int(pcbnew.Version()[0])
        if v == 7:
            for key in keys:
                if f.HasProperty(key):
                    return f.GetProperty(key)
        else:
            for key in keys:
                if f.HasField(key):
                    return f.GetFieldByName(key).GetText()
```

a-allard, 2024-02-28:
Tested on 8.0.0 and 7.0.11 on Windows 11.  Works like a charm on both.

pcbway, 2024-02-28:
We have submitted the code update, and when it is passed,the kicad 8 and kicad 7 will both work well.

pcbway, 2024-02-29:
Hi,Everyone,The plug-in has been updated and can now adapt to kicad8 and kicad7 versions. You can update and use it directly.
Thanks very much