pcbway/PCBWay-Plug-in-for-Kicad #18: Please update the plug for the KiCAD8.0 version.

yangtaogit, 2024-02-28:
The error under KiCAD8.0:  
[image]

Thanks a lot!

pcbway, 2024-02-29:
Hi,The plug-in has been updated and can now adapt to kicad8 and kicad7 versions. You can update and use it directly. Thanks very much

robouden, 2024-02-29:
In the nightly build, it works great!!!!!
So easy to order. What a blessing.

Regards
Rob Oudendijk

pcbway, 2024-02-29:
Glad to hear you like the plugin,Rob. We may hold Kicad design contest in the next month,hope you will take part in that time

robouden, 2024-02-29:
Please keep us informed.

Regards
Rob Oudendijk

pcbway, 2024-04-17:
> Hello, I am having a similar issue. Sometimes it works on projects, sometimes I get this error. I miss being able to use this plugin. ![image](https://private-user-images.githubusercontent.com/17507246/322960411-28fe6329-0b6b-4ca9-9e8e-9224f1ec94b4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzMjA5NjgsIm5iZiI6MTcxMzMyMDY2OCwicGF0aCI6Ii8xNzUwNzI0Ni8zMjI5NjA0MTEtMjhmZTYzMjktMGI2Yi00Y2E5LTllOGUtOTIyNGYxZWM5NGI0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE3VDAyMjQyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJlY2JhOTQ3YjA2OWI5YTNjZDc5MjJhN2E2ZWNlNjY1ZWZiMTc2ODE4MTZjZDdmMzIxNTEzYjMwNGRlOTAwMTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7N8CDb2SWc6I_KwtMkuufrvDmqqhwDPunhwJnvpY5FU)

Did you use the kicad 8 version plugin?

Joshc17m, 2024-04-17:
> Did you use the kicad 8 version plugin?

Yes, I'm using KiCAD 8 version. 
[image]

pcbway, 2024-04-18:
> > > > Hello, I am having a similar issue. Sometimes it works on projects, sometimes I get this error. I miss being able to use this plugin. ![image](https://private-user-images.githubusercontent.com/17507246/322960411-28fe6329-0b6b-4ca9-9e8e-9224f1ec94b4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzMjA5NjgsIm5iZiI6MTcxMzMyMDY2OCwicGF0aCI6Ii8xNzUwNzI0Ni8zMjI5NjA0MTEtMjhmZTYzMjktMGI2Yi00Y2E5LTllOGUtOTIyNGYxZWM5NGI0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE3VDAyMjQyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJlY2JhOTQ3YjA2OWI5YTNjZDc5MjJhN2E2ZWNlNjY1ZWZiMTc2ODE4MTZjZDdmMzIxNTEzYjMwNGRlOTAwMTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7N8CDb2SWc6I_KwtMkuufrvDmqqhwDPunhwJnvpY5FU)
> > > 
> > > 
> > > Did you use the kicad 8 version plugin?
> > 
> > 
> > Yes, I'm using KiCAD 8 version. ![image](https://private-user-images.githubusercontent.com/17507246/323223622-21f1eb1b-0a2c-4e2c-8d99-561578e62887.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzNzUxOTgsIm5iZiI6MTcxMzM3NDg5OCwicGF0aCI6Ii8xNzUwNzI0Ni8zMjMyMjM2MjItMjFmMWViMWItMGEyYy00ZTJjLThkOTktNTYxNTc4ZTYyODg3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE3VDE3MjgxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE1N2JiYWI1YTU5YjQ3OTg1NGY4ODZkMjg2MGZmYzA0YmI3YzViODE3YmQ4MTZiZWJiMjA2YWUyNGRmZTUxYjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.oNMc2FGXzvtHIrT9RnDk6C92UnQR1CuSjr3mD8ZxrsI)
> 
> Sorry, I realized I wasn't yet updated, just downloaded. I updated and now works great. Thank you for making me look further lol.

Glad to hear the problem solved,hope you can take part in our kicad design contest.

pcbway, 2024-07-11:
Hi,Guys,We have updated he latest plugin,it will get all component information,it will be more convenient.