pcbway/PCBWay-Plug-in-for-Kicad #20: does not appear in kicad 7 in ubuntu 22.04

davidedelvento, 2024-06-30:
Installing the plugin either from the zip downloaded from here, or from the list in the `Plugin and Content Manager` does absolutely nothing.
Going to the `Preferences->PCBEditor->Action Plugin` in the `PCB Editor` yields an empty list

I could find no way to have that green PCB Way button in the toolbar, which is shown in the gif at https://www.pcbway.com/blog/News/PCBWay_Plug_In_for_KiCad_3ea6219c.html

This is installing on from the official PPA

```
Application: KiCad x86_64 on x86_64

Version: 7.0.11-7.0.11~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.11.1
	HarfBuzz 6.0.0
	FontConfig 2.13.1
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.17

Platform: Ubuntu 22.04.4 LTS, 64 bit, Little endian, wxGTK, mate, x11

Build Info:
	Date: Feb 21 2024 19:16:04
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.6.3
	Curl: 7.88.1
	ngspice: 38
	Compiler: GCC 11.4.0 with C++ ABI 1016

Build settings:
	KICAD_SPICE=ON
```

pcbway, 2024-07-02:
Hi
I suggest you could try the Kicad version 8. Could you send mail to anson@pcbway.com. He will help solve the problem

pcbway, 2024-07-11:
We have updated this issue on the latest plugin,please double check it.