pcbway/PCBWay-Plug-in-for-Kicad #22: Components like fidicials, symbols, and other components marked as exclude from BOM being included in BOM

aaronw2, 2024-07-11:
In the last few boards I have submitted to PCBWay it is including things like fidicials, netties, mounting holes, testpoints, and other items that should not be included in the BOM. In fact they have the attribute set "Exclude from bill of materials" in the footprint.
These need to be honored by the plugin since I kept getting questioned as to if they're not to be included or not.

pcbway, 2024-07-11:
Hi,We have just updated the latest version of the kicad plug-in, which can obtain all component information. You could try it.
If you still meet problems,please send mail to anson@pcbway.com
Regards

jasen-b, 2024-07-28:
here's a fix: in thread.py after "for i, f in enumerate(footprints):"
add "if f.HasProperty('exclude_from_bom') : continue"
(as this is python code this should be indented using spaces to the save level as the next line)

chibiconsulting, 2024-09-02:
Using KiCad 8.0.4, and pcbway plugin v1.0.0, the items marked exclude from BOM are still included. This is causing time lost in questions about parts not required.

pcbway, 2025-01-16:
Could you send the file to anson@pcbway.com? We will check it