pcbway/PCBWay-Plug-in-for-Kicad #23: Generated BOM misses some Part Numbers

sp5wwp, 2024-11-23:
After generating the Bill of Materials for my PCB and placing an order, I received an email stating that my BOM lacks certain PNs. All of the values were set in the schematic. I double checked if anything is missing, yet still 11 per 43 items had missing PN/MPN fields in the generated BOM.

pcbway, 2024-11-25:
Hi
Thanks for the information.Did you use the latest plugin? It include all the information of your file.
I will ask our colleague Anson to contact with you about the details

sp5wwp, 2024-11-25:
Yes, I used the version provided by KiCAD, which seems to be the latest.

pcbway, 2025-01-16:
Could you send the file to anson@pcbway.com? We will check the reason