pcbway/PCBWay-Plug-in-for-Kicad #25: plugin for kicad 9.0 seems not to work

jwillner, 2025-03-31:
this plugin doesn't appear in the list. 
manuel install fails as well

pcbway, 2025-04-01:
> this plugin doesn't appear in the list. manuel install fails as well

The plugin now in the "manufacturer list"

pcbway, 2025-04-01:
We will test the manuel install function soon. There is no problem with internal installation.

pcbway, 2025-04-01:
Hi,Guys,we just tried manuel install and successed,could you send the video to me about how it failed? My mail is anson@pcbway.com

k4a-l, 2025-04-16:
@pcbway Manual install succeeded with kicad 9.0. But Installation from manager fails.
@asib The zip must use [Releases](https://github.com/pcbway/PCBWay-Plug-in-for-Kicad/releases). Not `Source code`.

pcbway, 2025-04-17:
> [@k4a-l](https://github.com/k4a-l) Aha, the readme says:
> 
> ```
> You can also download the latest ZIP file from https://github.com/pcbway/PCBWay-Plug-in-for-Kicad
> ```
> 
> [@pcbway](https://github.com/pcbway) Might be worth updating the link quoted above to point directly to the latest release zip: https://github.com/pcbway/PCBWay-Plug-in-for-Kicad/releases/download/v1.0.2/KiCadToPCBWay_v1.0.2.zip

Thanks,I will update it